UCSolarCarTeam/Epsilon-Telemetry-Server #32: State of Charge right side tab not working

Reddragoneer1, 2018-11-24:


bill-luu, 2018-11-26:
More description pls :)

bill-luu, 2018-11-27:
When you say functions, does that mean stuff is changing?

Sounds like it's just always off

henrytran28, 2018-11-28:
So when I just tested it with dummy values it worked fine. But because the server part never got implemented it always stays at the same value.

Reddragoneer1, 2018-11-28:
Sure, that'd be great

henrytran28, 2018-12-02:
So it was working right? I think that even though its duplicated in the Battery page the other pages don't have it and it might be useful to see it but that't up to you to decide.

bill-luu, 2018-12-13:
Does it work then? If it doesn't, we should take some time to see if it's actually going to be a difficult fix, if it's just a matter of a small mistake, we might as well fix it. If it requires a ton of changes we can get rid of it. 

It'll be convenient to have just there as a quick reference.

Also for the record, I think page loading isn't really affected by internet speed, since this is a single page application (The entire website is loaded when we access it basically). But if there's bad internet, yeah we won't be getting updates.

closed 2019-01-23 by bill-luu