UCSolarCarTeam/Epsilon-Telemetry-Server #136: Reading as int instead of big int

Reddragoneer1, 2019-06-23:
When running npm run start with new lap data time format

Console Log:

red@red-VirtualBox ~/SolarCar/Epsilon-Telemetry-Server/server $ npm run start

> epsilon-backend@0.0.0 start /home/red/SolarCar/Epsilon-Telemetry-Server/server
> node ./bin/www

Updating column names...
QueryResultError: No data returned from the query.
    at Query.ctx.db.client.query (/home/red/SolarCar/Epsilon-Telemetry-Server/server/node_modules/pg-promise/lib/query.js:209:41)
    at Query.handleReadyForQuery (/home/red/SolarCar/Epsilon-Telemetry-Server/server/node_modules/pg/lib/query.js:125:10)
    at Connection.<anonymous> (/home/red/SolarCar/Epsilon-Telemetry-Server/server/node_modules/pg/lib/client.js:227:19)
    at Connection.emit (events.js:182:13)
    at Socket.<anonymous> (/home/red/SolarCar/Epsilon-Telemetry-Server/server/node_modules/pg/lib/connection.js:126:12)
    at Socket.emit (events.js:182:13)
    at addChunk (_stream_readable.js:283:12)
    at readableAddChunk (_stream_readable.js:264:11)
    at Socket.Readable.push (_stream_readable.js:219:10)
    at TCP.onStreamRead [as onread] (internal/stream_base_commons.js:94:17)
ERROR: The previous query failed to execute.
Column names updated
Express: waiting for messages in msgQueue
^[[A
1 row inserted from RabbitMQ
1 row inserted from RabbitMQ
1 row inserted from RabbitMQ
1 row inserted from RabbitMQ
1 row inserted from RabbitMQ
error: value "1561316360615" is out of range for type integer
    at Connection.parseE (/home/red/SolarCar/Epsilon-Telemetry-Server/server/node_modules/pg/lib/connection.js:602:11)
    at Connection.parseMessage (/home/red/SolarCar/Epsilon-Telemetry-Server/server/node_modules/pg/lib/connection.js:399:19)
    at Socket.<anonymous> (/home/red/SolarCar/Epsilon-Telemetry-Server/server/node_modules/pg/lib/connection.js:121:22)
    at Socket.emit (events.js:182:13)
    at addChunk (_stream_readable.js:283:12)
    at readableAddChunk (_stream_readable.js:264:11)
    at Socket.Readable.push (_stream_readable.js:219:10)
    at TCP.onStreamRead [as onread] (internal/stream_base_commons.js:94:17)
ERROR: The previous query failed to execute.
1 row inserted from RabbitMQ
1 row inserted from RabbitMQ
1 row inserted into Lap Table
1 row inserted from RabbitMQ
1 row inserted from RabbitMQ
1 row inserted from RabbitMQ
1 row inserted from RabbitMQ
1 row inserted into Lap Table

Mavestri, 2019-06-24:
Not sure why this is happening, JavaScript should support numbers up to 2^53 and this is within the range 2^40-2^41. I'll look into it during tomorrow's meeting

Mavestri, 2019-06-27:
#137 should have fixed this, reopen if this occurs again

PR #137 (2019-06-27): Fix secondsdifference returning a bigint
Fixes #136
closed 2019-06-27 by Mavestri